easyeda/easyeda-api-sdk #2: Dead link?

adammakesthingsdev, 2024-07-31:
Firstly, thank you very much for development on this project, I am very much looking forward to using the API for the pro version. However, the link in the readme to the english documentation for the API does not work and redirects to the main page.
Thanks for your time developing this!

yanranxiaoxi, 2024-07-31:
The English document is in the process of being translated, and I will metion you here when I finish the translation.

adammakesthingsdev, 2024-08-01:
Thank you very much! I will machine translate now to follow along, happily awaiting an official translation though.

yanranxiaoxi, 2024-08-03:
> Thank you very much! I will machine translate now to follow along, happily awaiting an official translation though.

The English version of the Guide is now available:
https://prodocs.lceda.cn/en/api/guide/

Unfortunately, however, there are no plans to translate the reference document into English at this time. So you may still need to resort to partial machine translation in order to view this section.