easyeda/easyeda-api-sdk #3: 是否可以提供一些使用例程呢?

itldg, 2024-09-20:
想自行开发一个扩展插件
可以实现元件和边框的绑定关系,自己计算开孔尺寸和距离等

希望在弹窗中可以实现一个类似图层管理一样,添加删除 盖板

选择某个盖板后可以选择某个元件和某个边框的绑定关系

目前下载了SDK可以正常编译导入

如何实现在弹窗中调用eda的内容?

弹窗中是否可以使用eda样式的文本,表格,tab等

yanranxiaoxi, 2024-09-20:
使用例程是我们开发的下一个目标，在有空余开发资源后会进行编写。
现在我们内部有使用扩展实现许多功能，后续会将这些官方扩展作为例程开源提供。

IFrame 中可以直接使用 eda 对象进行调用。

IFrame 当前是完整的独立 HTML 页面，而不是以控件的形式，可自定义性更高，但没有办法自动套用 EDA 默认的样式，这是我们对当前功能复杂度方面的取舍。

itldg, 2024-09-20:
我有测试,在IFrame 写代码

```
console.log('[DEBUG] eda:', eda);
```

 Uncaught ReferenceError: eda is not defined

itldg, 2024-09-21:
在框架中我未能获取到 eda 的对象

另外想咨询下,是否可以在PCB中获取 矩形图元

yanranxiaoxi, 2024-09-21:
> 我有测试,在IFrame 写代码
> 
> ```
> console.log('[DEBUG] eda:', eda);
> ```
> 
> Uncaught ReferenceError: eda is not defined

当前框架内的 eda 对象在初始化时并未加载，是在初始化之后的时刻加载的，
你可以尝试循环等待其加载后再开始后续调用

itldg, 2024-09-23:
> > 我有测试,在IFrame 写代码
> > ```
> > console.log('[DEBUG] eda:', eda);
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Uncaught ReferenceError: eda is not defined
> 
> 当前框架内的 eda 对象在初始化时并未加载，是在初始化之后的时刻加载的， 你可以尝试循环等待其加载后再开始后续调用

感谢指导,今天验证了下,获取正常了,接下来等一等 PCB中获取 矩形图元,预计这个要多久会发布API呢

yanranxiaoxi, 2024-10-15:
@itldg 要绘制矩形，你可以使用折线图元，为其传入包含矩形 TPCB_PolygonSourceArray 的 IPCB_Polygon 对象

Sheepfan0828, 2024-12-13:
> 使用例程是我们开发的下一个目标，在有空余开发资源后会进行编写。 现在我们内部有使用扩展实现许多功能，后续会将这些官方扩展作为例程开源提供。
> 
> IFrame 中可以直接使用 eda 对象进行调用。
> 
> IFrame 当前是完整的独立 HTML 页面，而不是以控件的形式，可自定义性更高，但没有办法自动套用 EDA 默认的样式，这是我们对当前功能复杂度方面的取舍。

问下IFRAME如何打开？

itldg, 2024-12-13:
> > 使用例程是我们开发的下一个目标，在有空余开发资源后会进行编写。 现在我们内部有使用扩展实现许多功能，后续会将这些官方扩展作为例程开源提供。
> > IFrame 中可以直接使用 eda 对象进行调用。
> > IFrame 当前是完整的独立 HTML 页面，而不是以控件的形式，可自定义性更高，但没有办法自动套用 EDA 默认的样式，这是我们对当前功能复杂度方面的取舍。
> 
> 问下IFRAME如何打开？

eda.sys_IFrame.openIFrame('/iframe/index.html', 500, 500);

Sheepfan0828, 2024-12-13:
> > > 使用例程是我们开发的下一个目标，在有空余开发资源后会进行编写。 现在我们内部有使用扩展实现许多功能，后续会将这些官方扩展作为例程开源提供。
> > > IFrame 中可以直接使用 eda 对象进行调用。
> > > IFrame 当前是完整的独立 HTML 页面，而不是以控件的形式，可自定义性更高，但没有办法自动套用 EDA 默认的样式，这是我们对当前功能复杂度方面的取舍。
> > 
> > 
> > 问下IFRAME如何打开？
> 
> eda.sys_IFrame.openIFrame('/iframe/index.html', 500, 500);
这个是在哪里运行呢？

itldg, 2024-12-13:
https://prodocs.lceda.cn/cn/api/guide/inline-frame.html
有文档可以学习的

Sheepfan0828, 2024-12-13:
> https://prodocs.lceda.cn/cn/api/guide/inline-frame.html 有文档可以学习的

我的意思是这个在哪个地方跑？是在教程里的入口代码文件里面吗？

Sheepfan0828, 2024-12-13:
我在入口文件.src/index.ts里面跑这个 没有办法获取eda实例 但是代码编写的时候先写eda然后后面加个‘.' 然后还能正常弹出eda里面的类。
[image]

> https://prodocs.lceda.cn/cn/api/guide/inline-frame.html 有文档可以学习的

itldg, 2024-12-13:
// eslint-disable-next-line @typescript-eslint/no-unused-vars
export function activate(status?: 'onStartupFinished', arg?: string): void {
	console.log(extensionConfig.displayName, "激活", status, arg);
}

写在这里呢

Sheepfan0828, 2024-12-13:
> // eslint-disable-next-line @typescript-eslint/no-unused-vars export function activate(status?: 'onStartupFinished', arg?: string): void { console.log(extensionConfig.displayName, "激活", status, arg); }
> 
> 写在这里呢

写在这里还是说找不到eda
```

/**
 * 入口文件
 *
 * 本文件为默认扩展入口文件，如果你想要配置其它文件作为入口文件�?
 * 请修�? `extension.json` 中的 `entry` 字段�?
 *
 * 请在此处使用 `export`  导出所有你希望�? `headerMenus` 中引用的方法�?
 * 方法通过方法名与 `headerMenus` 关联�?
 *
 * 如需了解更多开发细节，请阅读：
 * https://prodocs.lceda.cn/cn/api/guide/
 */
import * as extensionConfig from '../extension.json';

// eslint-disable-next-line @typescript-eslint/no-unused-vars
export function activate(status?: 'onStartupFinished', arg?: string): void {
	eda.sys_IFrame.openIFrame('/iframe/index.html', 500, 500);
}

export function about(): void {
	eda.sys_MessageBox.showInformationMessage(
		eda.sys_I18n.text('EasyEDA extension SDK v', undefined, undefined, extensionConfig.version),
		eda.sys_I18n.text('About'),
	);
}

```

yanranxiaoxi, 2024-12-23:
> 在 IFrame 是否可以引入其它JS和CSS呢?
> 
> 比如使用Vue编译的结果放在IFrame中是否可行呢?

现在扩展的加载逻辑已经调整。`eda` 的注入阶段已经在页面加载之前。

引用 JS 和 CSS 也已就绪。

itldg, 2024-12-23:
> > 在 IFrame 是否可以引入其它JS和CSS呢?
> > 比如使用Vue编译的结果放在IFrame中是否可行呢?
> 
> 现在扩展的加载逻辑已经调整。`eda` 的注入阶段已经在页面加载之前。
> 
> 引用 JS 和 CSS 也已就绪。

好的,有机会我再来试试

itldg, 2024-12-23:
> > 在 IFrame 是否可以引入其它JS和CSS呢?
> > 比如使用Vue编译的结果放在IFrame中是否可行呢?
> 
> 现在扩展的加载逻辑已经调整。`eda` 的注入阶段已经在页面加载之前。
> 
> 引用 JS 和 CSS 也已就绪。

已测试,很棒,可以引入Vue的编译结果了,框架如果去除边框会更美观一些

[image]

[image]

yanranxiaoxi, 2024-12-23:
16px 的边缘在新版本已经移除了

yanranxiaoxi, 2024-12-23:
奇怪，在 2.2.34.5 应该已经移除它了 ...

itldg, 2024-12-23:
[image]

这个单位是如何转换的呢?

如果用户输入1mm,需要eda.sys_Unit.mmToMil(1, 5)/10 这样去转换么?

另外就是线条是否可以直接改长度?矩形直接修改宽高?

easyeda2021, 2025-09-19:
你好，你这个扩展可以上架在我们扩展广场给大家使用吗

itldg, 2025-09-19:
> 你好，你这个扩展可以上架在我们扩展广场给大家使用吗

很遗憾当时的完成度很低,大部分靠手动. 但是最终生成的结果导出到CAD,再到立创铝型材开孔很方便

后续如果有机会的话我再看看现在的API,看看能否完成当时的想法