easyeda/easyeda-api-sdk #4: 类型标记错误?

itldg, 2024-09-21:
https://prodocs.lceda.cn/cn/api/reference/pro-api.ipcb_layeritem.html

type |   | EPCB_LayerType | 类型

[image]

实际获取到的是字符串类型

[image]

另外图层属性乱码,文件中还有许多这种乱码的存在

[image]

[image]

[image]

[image]

itldg, 2024-09-21:
`getSelectedPrimitives` 提示需要用 `SCH_SelectControl.getAllSelectedPrimitives`

`SCH_SelectControl.getAllSelectedPrimitives` 此版本被排除

[image]

yanranxiaoxi, 2024-09-21:
> `getSelectedPrimitives` 提示需要用 `SCH_SelectControl.getAllSelectedPrimitives`
>
> `SCH_SelectControl.getAllSelectedPrimitives` 此版本被排除 [...]

该接口存在 BUG 正在修复中，所以当前暂未提供

yanranxiaoxi, 2024-09-21:
乱码问题已定位，下一次版本推送将解决

itldg, 2024-09-25:
短期内,API 有望更新不,比如增加获取矩形,圆形等

yanranxiaoxi, 2024-10-15:
> 短期内,API 有望更新不,比如增加获取矩形,圆形等

圆形和矩形均可以使用折线图元绘制，只需为其传入包含矩形和圆形的 IPCB_Polygon

itldg, 2024-10-15:
获取矩形,圆形呢?PCB上已经有了,想自己做绑定关系,选择一个矩形,选择一个元件,设置偏移和尺寸,然后修改矩形的宽高和坐标

yanranxiaoxi, 2024-10-29:
> 获取矩形,圆形呢?PCB上已经有了,想自己做绑定关系,选择一个矩形,选择一个元件,设置偏移和尺寸,然后修改矩形的宽高和坐标

矩形和圆形均在 IPCB_Polygon 内有数据格式定义

easyeda2021, 2024-12-28:
2.2.34.8已经修复

closed 2025-02-28 by yanranxiaoxi