easyeda/easyeda-api-sdk #5: LIB_Device.search() 方法返回数据异常

klxf, 2024-10-26:
LIB_Device.search() 返回的搜索结果与传入的关键词**毫不相关**，搜索任何关键词都会返回相同结果

按照文档，下面这段代码可以从系统库中搜索“AMS1117-3.3”
```js
let key = "AMS1117-3.3";
eda.lib_LibrariesList.getSystemLibraryUuid().then((libraryUuid) => {
    eda.lib_Device.search(key, libraryUuid).then((result) => {
        console.log(result);
    });
});
```

输出结果却与关键词毫无关联：
<details><summary>返回数据</summary>
<p>

```json
[
    {
        "uuid": "bf0572b37d5747af8419bbbd11533598",
        "libraryUuid": "",
        "ordinal": 0,
        "name": "LM2576HVS-12_C18197844",
        "symbolName": "lm2576hvs-12_c18197844-1",
        "symbolUuid": "940891765d3e4cfa8767f110bda7d253",
        "footprintName": "PG-TO-263-5_L10.0-W9.0-P1.70-LS15.0-BR",
        "footprintUuid": "c6ba23d9041045508b5ab02b28994be6",
        "model3DName": "PG-TO-263-5_L10.0-W9.0-P1.70-LS15.0-BR",
        "model3DUuid": "a4dcf1cc949c44cb90ba861f2b9c6108",
        "imageUuid": "https://alimg.szlcsc.com/upload/public/product/middle/20231118/68A918BA4354205255F325D8EAD5D419.jpg",
        "description": "输入电压:-;输出电流:3A;",
        "manufacturer": "TECH PUBLIC(台舟)",
        "manufacturerId": "LM2576HVS-12",
        "supplier": "LCSC",
        "supplierId": "C18197844"
    },
    {
        "uuid": "23cd061a645a4bb3a48c68781de8e308",
        "libraryUuid": "",
        "ordinal": 1,
        "name": "R-78K5.0-0.5",
        "symbolName": "r-78k5.0-0.5",
        "symbolUuid": "56c9375012544f47b5f5f38cc6d65a49",
        "footprintName": "PWRM-TH_3P-L11.5-W7.6-P2.54_R-785.0-0.5",
        "footprintUuid": "c00e4f9c582547389aa2b632e552e5f3",
        "description": "",
        "manufacturer": "Recom Power",
        "manufacturerId": "R-78K5.0-0.5",
        "supplier": "LCSC",
        "supplierId": "C5716027"
    },
    {
        "uuid": "68466bb9e71c44b3ba5d2f998e1a0c71",
        "libraryUuid": "",
        "ordinal": 2,
        "name": "FEXL0530A-2R2M",
        "symbolName": "fexl0530a-2r2m",
        "symbolUuid": "c8f4e58d45c24b9d991855fb56dbafcf",
        "footprintName": "IND-SMD_L5.5-W5.3_ECS-HCMPI-0503Q",
        "footprintUuid": "651685a42d9e48189493c5acdfbf0cb0",
        "imageUuid": "https://alimg.szlcsc.com/upload/public/product/middle/20230404/9876099359E2635FF34F350D18FE3165.jpg",
        "description": "电感值:2.2uH;精度:±20%;额定电流:9.7A;饱和电流(Isat):10A;类型:一体成型电感;",
        "manufacturer": "cjiang(长江微电)",
        "manufacturerId": "FEXL0530A-2R2M",
        "supplier": "LCSC",
        "supplierId": "C5378511"
    },
    {
        "uuid": "2099d46879164f4a9d0d9ec37882684e",
        "libraryUuid": "",
        "ordinal": 3,
        "name": "XT25F256BWSIGT",
        "symbolName": "xt25f256bwsigt",
        "symbolUuid": "a5de071a34574bb19190162d8d84343d",
        "footprintName": "WSON-8_L8.0-W6.0-P1.27-BL-EP-2",
        "footprintUuid": "4336e2cba1d544a6aa6b80a9444a5e82",
        "imageUuid": "https://alimg.szlcsc.com/upload/public/product/middle/20230107/3CD5997AB65F7238EE2C08826F4DC728.jpg",
        "description": "接口类型:SPI;存储容量:256Mbit;时钟频率(fc):120MHz;",
        "manufacturer": "XTX(芯天下)",
        "manufacturerId": "XT25F256BWSIGT",
        "supplier": "LCSC",
        "supplierId": "C2759959"
    },
    {
        "uuid": "a5dda7fd49a14f26894d3338802674a4",
        "libraryUuid": "",
        "ordinal": 4,
        "name": "LTM8031IV#PBF",
        "symbolName": "ltm8031iv#pbf",
        "symbolUuid": "798d77c1d93347f18c111599632d8259",
        "footprintName": "LGA-71_L15.0-W9.0-P1.27-TL_ADI_LTM8032XXX",
        "footprintUuid": "02c0fabc3f534d409b36215a2f28929c",
        "model3DName": "LGA-71_L15.0-W9.0-P1.27-TL_ADI_LTM8032XXX",
        "model3DUuid": "4b7a1de1455a47bc9cc35703de6735da",
        "description": "输入电压:3.6V~36V;输出电流:1A;",
        "manufacturer": "ADI(亚德诺)",
        "manufacturerId": "LTM8031IV#PBF",
        "supplier": "LCSC",
        "supplierId": "C690668"
    },
    {
        "uuid": "7b13f40036944020918082656df923e5",
        "libraryUuid": "",
        "ordinal": 5,
        "name": "HEF4528BP652",
        "symbolName": "hef4528bp652",
        "symbolUuid": "88cb40e85a7f4841a72e7a5a4728731f",
        "footprintName": "PDIP-16_L19.3-W6.4-P2.54-LS7.94-BL",
        "footprintUuid": "b24b4737a74b48fb91593464cdf5e525",
        "model3DName": "PDIP-16_L19.3-W6.4-P2.54-LS7.94-BL",
        "model3DUuid": "96ff5d01b3074dd183bcbc4cd00a63de",
        "description": "",
        "manufacturer": "Nexperia(安世)",
        "manufacturerId": "HEF4528BP652",
        "supplier": "LCSC",
        "supplierId": "C3236168"
    },
    {
        "uuid": "772295220f114fe5b845643420a52277",
        "libraryUuid": "",
        "ordinal": 6,
        "name": "TD321SCAN",
        "symbolName": "TD321SCAN",
        "symbolUuid": "3ba13211713d4a67b6baaaf819519dbb",
        "footprintName": "COMM-SMD_7P-L18.2-W14.8-P2.54_TD321SCAN",
        "footprintUuid": "66743d3e9df04b2dab7bc4d075b22cae",
        "description": "",
        "manufacturer": "MORNSUN(金升阳)",
        "manufacturerId": "TD321SCAN",
        "supplier": "LCSC",
        "supplierId": "C20615117"
    },
    {
        "uuid": "4db7a53fed00422d88bf7a5da9a1feac",
        "libraryUuid": "",
        "ordinal": 7,
        "name": "ESD8D12C",
        "symbolName": "ESD8D12C_C22368679",
        "symbolUuid": "ca4ebe228f224db98d1cdd2b5dd8d27c",
        "footprintName": "DFN1006-2L-BI",
        "footprintUuid": "4fb383539f354755a76680fd7c6083af",
        "model3DName": "DFN1006-2L-BI",
        "model3DUuid": "9c805dabf26842999d981f2745203d13",
        "description": "",
        "manufacturer": "Wild Goose(威谷)",
        "manufacturerId": "ESD8D12C",
        "supplier": "LCSC",
        "supplierId": "C22368679"
    },
    {
        "uuid": "44ce9ed0198d414094cc7300e6ffdb4c",
        "libraryUuid": "",
        "ordinal": 8,
        "name": "CC6430SS",
        "symbolName": "CC6430SS",
        "symbolUuid": "a7c8bde8c6a9410d9692bfb8f5773208",
        "footprintName": "SOT-335_L2.8-W2.6-BL_CC6430SS",
        "footprintUuid": "95b3134d80fb4fe8bb88c9648677e880",
        "imageUuid": "https://alimg.szlcsc.com/upload/public/product/middle/20241021/4A36429A9FB5353BED89D1CEC84A24AC.jpg",
        "description": "",
        "manufacturer": "CrossChip(成都芯进)",
        "manufacturerId": "CC6430SS",
        "supplier": "LCSC",
        "supplierId": "C22391864"
    },
    {
        "uuid": "ef29e1716c704641822f2df1e357e69d",
        "libraryUuid": "",
        "ordinal": 9,
        "name": "LD-SM-430",
        "symbolName": "LD-SM-430",
        "symbolUuid": "dcc3731eac9043a2bf8ab2959b1767e4",
        "footprintName": "SMD_LD-SM-430",
        "footprintUuid": "89708dc3031e447e81a45eec14f406f0",
        "imageUuid": "https://alimg.szlcsc.com/upload/public/product/middle/20240822/6BBF2EAF94229915295F64CA7E67C24A.jpg",
        "description": "贴片马达430",
        "manufacturer": "LEADER(立得)",
        "manufacturerId": "LD-SM-430",
        "supplier": "LCSC",
        "supplierId": "C41348533"
    }
]
```

</p>
</details>

一点猜测：
大概是 `https://client/api/v2/devices` 这个 api 的问题？相同参数请求 `https://client/api/devices/search` 是可以拿到正确结果的

yanranxiaoxi, 2024-11-01:
在线上和内部环境均未发现该问题，可能是搜索服务器突发异常，你现在再看一下

klxf, 2024-11-01:
刚刚试了一下，仍然存在这个问题
[image]

yanranxiaoxi, 2024-11-01:
客户端环境已记录

yanranxiaoxi, 2024-11-06:
已修复到客户端半离线模式

closed 2024-11-06 by yanranxiaoxi