easyeda/easyeda-api-sdk #6: SCH_PrimitiveComponent.create() 方法存在问题

klxf, 2024-10-28:
### 参数问题
按照文档传入了必要参数，但是却出现报错
```js
componentUUID = "9f9c6cb41c7449fd8acf96aceed2661a";
subLibraryId = "0819f05c4eef4c71ace90d822a990e87";
x = 200;
y = 500;

eda.sch_PrimitiveComponent.create(componentUUID, subLibraryId, x, y).then((result) => {
	console.log(result);
});
```
```log
Uncaught (in promise) Error: 数据不符合规范
    at de.setState_AddIntoBom (api.js:4:144858)
    at de.create (api.js:4:142431)
    at Ai.create (api.js:4:137690)
    at a0560074-4ff6-4097-8360-9b6d208010ba:370:33
```

### 坐标问题
传入所有参数能放置元件，但是放置的坐标存在问题，下面的代码将在 `200, 500` 放置元件，但是实际却放置在了` 200, -200`（最神奇的是使用 `getState_X`, `getState_Y` 获取到的坐标还是正确的 `200, 500`）
```js
componentUUID = "9f9c6cb41c7449fd8acf96aceed2661a";
subLibraryId = "0819f05c4eef4c71ace90d822a990e87";
x = 200;
y = 500;
rotation = 0;
mirror = false;
libraryPath = "0819f05c4eef4c71ace90d822a990e87";
addIntoBom = true;
addIntoPcb = true;
eda.sch_PrimitiveComponent.create(componentUUID, subLibraryId, x, y, rotation, mirror, libraryPath, addIntoBom, addIntoPcb).then((result) => {
	console.log(result.getState_X() + ", " + result.getState_Y());
});

yanranxiaoxi, 2024-10-29:
该问题已建单，等待修复中~

yanranxiaoxi, 2024-12-18:
在 2.2.34 版本已修复

closed 2024-12-18 by yanranxiaoxi