easyeda/easyeda-api-sdk #7: 请求新增“获取鼠标在画布的坐标”和“放置元件”接口

klxf, 2024-10-30:
### 1. 获取鼠标在画布的坐标
如题，获取鼠标在画布的坐标

### 2. 放置元件
目前的 `create` 方法需要指定一个坐标才能放置元件，
请求新增一个和如下功能相同的接口：传入 `componentUuid` 和 `subLibraryId`，将对应器件符号吸附到鼠标指针，左键放置、右键取消...
[image]

yanranxiaoxi, 2024-10-31:
1. 获取鼠标在画布的坐标接口已经设计等待开发
2. 将待放置的器件吸附到鼠标功能希望能得到具体的使用场景反馈，即在什么场景下会用到这个

klxf, 2024-10-31:
> 2. 将待放置的器件吸附到鼠标功能希望能得到具体的使用场景反馈，即在什么场景下会用到这个

目前想实现在 iframe 内选择一个器件，可以由用户自由放置（类似“器件/复用模块”窗口那样）的功能

（这个功能也可以用之前提到的两个需求 iframe 消息传递+获取坐标来实现

yanranxiaoxi, 2024-11-01:
看起来是一个确实可能会用到的功能，准备提供两个接口（原理图+PCB）

Sheepfan0828, 2024-12-19:
> ### 1. 获取鼠标在画布的坐标
> 如题，获取鼠标在画布的坐标
> 
> ### 2. 放置元件
> 目前的 `create` 方法需要指定一个坐标才能放置元件， 请求新增一个和如下功能相同的接口：传入 `componentUuid` 和 `subLibraryId`，将对应器件符号吸附到鼠标指针，左键放置、右键取消... ![QQ_1730278707139](https://private-user-images.githubusercontent.com/31070597/381466725-ce1ecdf1-9bf6-4f24-8849-8654c2d4b0e0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ1Nzc5NDEsIm5iZiI6MTczNDU3NzY0MSwicGF0aCI6Ii8zMTA3MDU5Ny8zODE0NjY3MjUtY2UxZWNkZjEtOWJmNi00ZjI0LTg4NDktODY1NGMyZDRiMGUwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE5VDAzMDcyMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI2NDA2MmRhNGNhZTcwZjdlMTY0YzMwNmE0ZDAwNGFhZjc5ZWU3OGE4ZjlhMzRiYjRiMDgzOWJhYTVmOTFkM2EmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.0tpYCVSwdotESptidEHD4Y_DlsYeJpW_q-ENQY-gm6Q)

你好！请问如何使用`create`方法放置原件呀？

klxf, 2024-12-21:
> 你好！请问如何使用`create`方法放置原件呀？

https://prodocs.lceda.cn/cn/api/reference/pro-api.sch_primitivecomponent.create.html

像下面这样，`libraryUuid` 可以用 [LIB_LibrariesList](https://prodocs.lceda.cn/cn/api/reference/pro-api.lib_librarieslist.html) 类中的方法获得，元件的 `uuid` 可以用 [LIB_Device](https://prodocs.lceda.cn/cn/api/reference/pro-api.lib_device.html) 类中的方法获得 

```ts
libraryUUID = "0819f05c4eef4c71ace90d822a990e87";
componentUUID = "9f9c6cb41c7449fd8acf96aceed2661a";
x = 100;
y = 100;
eda.sch_PrimitiveComponent.create({"libraryUuid": libraryUUID , "uuid": componentUUID , x, y);
```

easyeda2021, 2024-12-28:
在最新的2.2.34.8已经支持获取鼠标位置接口

klxf, 2024-12-29:
> 在最新的2.2.34.8已经支持获取鼠标位置接口

🤔 文档有更新吗
没找到接口文档，请问可以给个链接吗

closed 2025-06-16 by yanranxiaoxi