easyeda/easyeda-api-sdk #8: 如何进行API调用

Sheepfan0828, 2024-12-13:
[image]
你好，请问下这上面的代码是在哪里跑呢？然后如何将这些API调用与嘉立创EDA建立联系呢？

yanranxiaoxi, 2024-12-18:
请查阅文档的如何开始部分：https://prodocs.lceda.cn/cn/api/guide/how-to-start.html

扩展需要打包后导入嘉立创 EDA 专业版调用，而不是使用 ts-node

Sheepfan0828, 2024-12-18:
> 请查阅文档的如何开始部分：https://prodocs.lceda.cn/cn/api/guide/how-to-start.html
> 
> 扩展需要打包后导入嘉立创 EDA 专业版调用，而不是使用 ts-node

请问如何打包呢？

yanranxiaoxi, 2024-12-18:
https://prodocs.lceda.cn/cn/api/guide/how-to-start.html#v-%E6%9E%84%E5%BB%BA

打包完成后，将其导入嘉立创 EDA 专业版中：`设置` -> `扩展` -> `扩展管理器`

Sheepfan0828, 2024-12-18:
> https://prodocs.lceda.cn/cn/api/guide/how-to-start.html#v-%E6%9E%84%E5%BB%BA
> 
> 打包完成后，将其导入嘉立创 EDA 专业版中：`设置` -> `扩展` -> `扩展管理器`

我把放在`/build/dist/`目录下的文件按照这个步骤弄了 但是扩展导入后 没有反应 以下是我在`.src/index.ts`的代码：
```
/**
 * 入口文件
 *
 * 本文件为默认扩展入口文件，如果你想要配置其它文件作为入口文件�?
 * 请修�? `extension.json` 中的 `entry` 字段�?
 *
 * 请在此处使用 `export`  导出所有你希望�? `headerMenus` 中引用的方法�?
 * 方法通过方法名与 `headerMenus` 关联�?
 *
 * 如需了解更多开发细节，请阅读：
 * https://prodocs.lceda.cn/cn/api/guide/
 */
import * as extensionConfig from '../extension.json';

// eslint-disable-next-line @typescript-eslint/no-unused-vars
export function activate(status?: 'onStartupFinished', arg?: string): void {
	eda.sys_IFrame.openIFrame('/iframe/index.html', 500, 500);
}

export function about(): void {
	eda.sys_MessageBox.showInformationMessage(
		eda.sys_I18n.text('EasyEDA extension SDK v', undefined, undefined, extensionConfig.version),
		eda.sys_I18n.text('About'),
	);
}
```