easyeda/easyeda-api-sdk #9: 无法找到eda

Sheepfan0828, 2024-12-13:
在编写ts的时候，我可以正常引用`eda`，而且编写的时候也能够让它调用其中的实例和方法，但是跑的时候报错，说找不到`eda`。
如下是`.src/index.ts`代码 在`sdk`项目里面
```
/**
 * 入口文件
 *
 * 本文件为默认扩展入口文件，如果你想要配置其它文件作为入口文件�?
 * 请修�? `extension.json` 中的 `entry` 字段�?
 *
 * 请在此处使用 `export`  导出所有你希望�? `headerMenus` 中引用的方法�?
 * 方法通过方法名与 `headerMenus` 关联�?
 *
 * 如需了解更多开发细节，请阅读：
 * https://prodocs.lceda.cn/cn/api/guide/
 */
import * as extensionConfig from '../extension.json';

// eslint-disable-next-line @typescript-eslint/no-unused-vars
export function activate(status?: 'onStartupFinished', arg?: string): void {
	console.log('[DEBUG] eda:', eda);
	eda.sys_IFrame.openIFrame('/iframe/index.html', 500, 500);
}

export function about(): void {
	eda.sys_MessageBox.showInformationMessage(
		eda.sys_I18n.text('EasyEDA extension SDK v', undefined, undefined, extensionConfig.version),
		eda.sys_I18n.text('About'),
	);
}
```
下面是运行之后产生的错误信息
```
$ npx ts-node ./src/index.ts
D:\jlceda-extension\pro-api-sdk\node_modules\ts-node\src\index.ts:859
    return new TSError(diagnosticText, diagnosticCodes, diagnostics);
           ^
TSError: ⨯ Unable to compile TypeScript:
src/index.ts:17:2 - error TS2304: Cannot find name 'eda'.

17  eda.sys_IFrame.openIFrame('/iframe/index.html', 500, 500);
    ~~~
src/index.ts:21:2 - error TS2304: Cannot find name 'eda'.

21  eda.sys_MessageBox.showInformationMessage(
    ~~~
src/index.ts:22:3 - error TS2304: Cannot find name 'eda'.

22   eda.sys_I18n.text('EasyEDA extension SDK v', undefined, undefined, extensionConfig.version),
     ~~~
src/index.ts:23:3 - error TS2304: Cannot find name 'eda'.

23   eda.sys_I18n.text('About'),
     ~~~

    at createTSError (D:\jlceda-extension\pro-api-sdk\node_modules\ts-node\src\index.ts:859:12)
    at reportTSError (D:\jlceda-extension\pro-api-sdk\node_modules\ts-node\src\index.ts:863:19)
    at getOutput (D:\jlceda-extension\pro-api-sdk\node_modules\ts-node\src\index.ts:1077:36)
    at Object.compile (D:\jlceda-extension\pro-api-sdk\node_modules\ts-node\src\index.ts:1433:41)
    at Module.m._compile (D:\jlceda-extension\pro-api-sdk\node_modules\ts-node\src\index.ts:1617:30)
    at node:internal/modules/cjs/loader:1718:10
    at Object.require.extensions.<computed> [as .ts] (D:\jlceda-extension\pro-api-sdk\node_modules\ts-node\src\index.ts:1621:12)
    at Module.load (node:internal/modules/cjs/loader:1305:32)
    at Function._load (node:internal/modules/cjs/loader:1119:12)
    at TracingChannel.traceSync (node:diagnostics_channel:322:14) {
  diagnosticCodes: [ 2304, 2304, 2304, 2304 ]
}

```

Sheepfan0828, 2024-12-17:
你好 请问什么时候可以帮助我解决这个问题？我这边比较急

yanranxiaoxi, 2024-12-18:
重复内容 https://github.com/easyeda/pro-api-sdk/issues/8

closed 2024-12-18 by yanranxiaoxi