easyeda/easyeda-api-sdk #10: 器件创建api调用实例

Sheepfan0828, 2024-12-18:
你好 我这边想尝试利用api来放置单个器件 但是放置的时候总是报错 原因应该是`libraryUuid`没有办法获取 在`console`里面打印总是空。如下是我编写的程序 想放置一个电阻 能否帮我看看这种形式问题出在哪里？
```
/**
 * 入口文件
 *
 * 本文件为默认扩展入口文件，如果你想要配置其它文件作为入口文件�?
 * 请修�? `extension.json` 中的 `entry` 字段�?
 *
 * 请在此处使用 `export`  导出所有你希望�? `headerMenus` 中引用的方法�?
 * 方法通过方法名与 `headerMenus` 关联�?
 *
 * 如需了解更多开发细节，请阅读：
 * https://prodocs.lceda.cn/cn/api/guide/
 */
import * as extensionConfig from '../extension.json';

// eslint-disable-next-line @typescript-eslint/no-unused-vars
export function activate(status?: 'onStartupFinished', arg?: string): void {
    eda.sys_IFrame.openIFrame('/iframe/index.html', 500, 500);
}

export async function about(): Promise<void> {
        const comp_info = (await eda.lib_Device.search("Resistor")).shift();
        if (!comp_info) {
            throw new Error("Failed to create info");
        }

        //let library_uuid : string = await comp_info.libraryUuid;
        eda.dmt_Workspace.toggleToWorkspace();
        const library_uuid = await eda.lib_LibrariesList.getSystemLibraryUuid();
        if (!library_uuid) {
           throw new Error("Failed to get project library UUID");
        }
        let c_uuid : string = await comp_info.uuid;

        const comp : {
            libraryUuid:string
            uuid:string        
        } = {libraryUuid:library_uuid, uuid:c_uuid};

        console.log("libraryUuid: ", library_uuid, "uuid: ", c_uuid);
        
        const cr  = await eda.sch_PrimitiveComponent.create(comp, 0, 0);
        if (!cr) {
            throw new Error("Failed to create ISCH_PrimitiveComponent.");
        }
        
        const c: ISCH_PrimitiveComponent = cr;
        //console.log("Component Created");

        c.done();
}
```

radeeyate, 2024-12-18:
You can use Debug Mode for this: https://prodocs.easyeda.com/en/api/guide/invoke-apis.html#entering-debug-mode

Sheepfan0828, 2024-12-19:
> You can use Debug Mode for this: https://prodocs.easyeda.com/en/api/guide/invoke-apis.html#entering-debug-mode

Hi bro! Thanks for your response. As I have debugged in debug mode, I have got an error as follows:
```
Uncaught (in promise) Error: 数据不符合规范
    at de.setState_ComponentUuid (api.js:4:147310)
    at de.create (api.js:4:142137)
    at Ai.create (api.js:4:137690)
    at Object.about (eval at hr (api.js:7:67), <anonymous>:46:49)
```
And this problem caused by following piece of line from my original implementation (as posted above):
`const cr  = await eda.sch_PrimitiveComponent.create(comp, 0, 0);`
This maybe because of `libraryUuid` I have got from `const library_uuid = await eda.lib_LibrariesList.getSystemLibraryUuid();` is empty. But I am confused why is this happen. Can you notice anything wrong on code?

yanranxiaoxi, 2024-12-19:
Using your method, I was able to get the library_uuid and c_uuid with the following code:

```javascript
const comp_info = (await eda.lib_Device.search('Resistor')).shift();
const library_uuid = comp_info.libraryUuid;
if (!library_uuid) {
    throw new Error("Failed to get project library uuid!");
}
const c_uuid = comp_info.uuid;
console.log('library_uuid:', library_uuid, 'c_uuid:', c_uuid);
```

If you are still having problems, can you tell me a bit about your runtime environment?

eg: your software version, whether you are on the browser side, whether you are in client offline mode...

Sheepfan0828, 2024-12-19:
> Using your method, I was able to get the library_uuid and c_uuid with the following code:
> 
> ```js
> const comp_info = (await eda.lib_Device.search('Resistor')).shift();
> const library_uuid = comp_info.libraryUuid;
> if (!library_uuid) {
>     throw new Error("Failed to get project library uuid!");
> }
> const c_uuid = comp_info.uuid;
> console.log('library_uuid:', library_uuid, 'c_uuid:', c_uuid);
> ```
> 
> If you are still having problems, can you tell me a bit about your runtime environment?
> 
> eg: your software version, whether you are on the browser side, whether you are in client offline mode...

Thanks for your response. Can you please be more specific about the software version? Like which software do you want？

Sheepfan0828, 2024-12-19:
> Using your method, I was able to get the library_uuid and c_uuid with the following code:
>
> ```js
> const comp_info = (await eda.lib_Device.search('Resistor')).shift();
> const library_uuid = [...]

Assuming you mean the EasyEDA version as the software version, so the version is v2.2.32 and I am on the browser site.

Sheepfan0828, 2024-12-20:
> Using your method, I was able to get the library_uuid and c_uuid with the following code:
>
> ```js
> const comp_info = (await eda.lib_Device.search('Resistor')).shift();
> const library_uuid = [...]

Hello! How is the things going? Really need the quick response here...

Sheepfan0828, 2024-12-20:
When using the method `const comp_info = await eda.lib_Device.search("r");`, the length of the list returned by this method is just 10 but I assume it should much more than 10 as we could have so many components in which their name involves the letter 'r'.

yanranxiaoxi, 2024-12-20:
I've gotten several emails prompting me that you've left a message here, but I don't see the message in the email every time I click in. Has the problem been resolved?

The new version is avaliable (2.2.34.5).

Please try to reproduce your problem in this release and provide the minimal reproduction code here.

Sheepfan0828, 2024-12-23:
> I've gotten several emails prompting me that you've left a message here, but I don't see the message in the email every time I click in. Has the problem been resolved?
>
> The new version is avaliable [...]

How to log in into this version on the browser site?

Sheepfan0828, 2024-12-23:
> > I've gotten several emails prompting me that you've left a message here, but I don't see the message in the email every time I click in. Has the problem been resolved?
> > The new version is avaliable (2.2.34.5).
> > Please try to reproduce your problem in this release and provide the minimal reproduction code here.
> 
> How to log in into this version on the browser site? And also, I have tried to use this piece of script to generate library uuid   `const library_uuid = await eda.lib_LibrariesList.getSystemLibraryUuid();` and I am able to get this uuid via this way but i don't know whether is this a correct way. Assuming this way is correct, I have got the following error when I calling the `create` method to generate the component. 
`Uncaught (in promise) Error: 数据不符合规范`
Here is my full version of the code, please take a look and try to run it on your machine to see whether it works. 
```
/**
 * 入口文件
 *
 * 本文件为默认扩展入口文件，如果你想要配置其它文件作为入口文件�?
 * 请修�? `extension.json` 中的 `entry` 字段�?
 *
 * 请在此处使用 `export`  导出所有你希望�? `headerMenus` 中引用的方法�?
 * 方法通过方法名与 `headerMenus` 关联�?
 *
 * 如需了解更多开发细节，请阅读：
 * https://prodocs.lceda.cn/cn/api/guide/
 */
import * as extensionConfig from '../extension.json';

// eslint-disable-next-line @typescript-eslint/no-unused-vars
export function activate(status?: 'onStartupFinished', arg?: string): void {
    eda.sys_IFrame.openIFrame('/iframe/index.html', 500, 500);
}

export async function about(): Promise<void> {
    const library_uuid = await eda.lib_LibrariesList.getSystemLibraryUuid();
    if (!library_uuid) {
        throw new Error("Failed to get project library uuid!");
    }
    const comp_info = (await eda.lib_Device.search("SM4007PL_C64898", library_uuid)).shift();
    if (!comp_info) {
        throw new Error("Failed to create info");
    }
    const c_uuid = await comp_info.uuid;

    //console.log("libraryUuid: ", library_uuid, "uuid: ", c_uuid, "uuid from comp_info: ", await comp_info.libraryUuid);

    let comp = {libraryUuid: library_uuid, uuid: c_uuid};

    const c  = await eda.sch_PrimitiveComponent.create(comp, 0, 0);
    if (!c) {
        throw new Error("Failed to create ISCH_PrimitiveComponent.");
    }
    console.log("Component Created");
    c.done();
}
```

yanranxiaoxi, 2024-12-23:
I tested your code at version 2.2.34.5 in browser and it worked fine without any errors.

[image]

yanranxiaoxi, 2024-12-23:
By the way, `c.done()` in the code is redundant. You can just remove it without affecting what your code does.

Sheepfan0828, 2024-12-23:
> I tested your code at version 2.2.34.5 in browser and it worked fine without any errors. [...]

That sounds good! How to switch the version on the browser site?

yanranxiaoxi, 2024-12-23:
Next to the version number in the top right corner of your page, there will be a small up arrow waiting for you.

Sheepfan0828, 2024-12-23:
> Next to the version number in the top right corner of your page, there will be a small up arrow waiting for you.

I cannot see any up arrows you mentioned...
[image]

yanranxiaoxi, 2024-12-23:
Okey... Looks like you may not have been pushed to.

You can use this address to access the current version: [https://pro.lceda.cn/2.2.34.6.9a154024/editor](https://pro.lceda.cn/2.2.34.6.9a154024/editor)

> [!WARNING]
> Please note: This is the address that specifies a fixed version (2.2.34.6), meaning that it will not be pushed to any newer version when it is in use.

Sheepfan0828, 2024-12-23:
> Okey... Looks like you may not have been pushed to.
>
> You can use this address to access the current version: https://pro.lceda.cn/2.2.34.6.9a154024/editor
>
> Warning
>
> Please note: This is the address [...]

Thank you! It works now:)

yanranxiaoxi, 2024-12-23:
If you want to output logs in the console, check out the documentation:

[https://prodocs.easyeda.com/en/api/guide/invoke-apis.html#entering-debug-mode](https://prodocs.easyeda.com/en/api/guide/invoke-apis.html#entering-debug-mode)

> add the `cll=debug` parameter within the URL of the EasyEDA Pro editor to enter debug mode.

Sheepfan0828, 2024-12-23:
> If you want to output logs in the console, check out the documentation:
> 
> https://prodocs.easyeda.com/en/api/guide/invoke-apis.html#entering-debug-mode
> 
> > add the `cll=debug` parameter within the URL of the EasyEDA Pro editor to enter debug mode.

Hello! Please see the new issue #13 I have posted. Thank you:)

Sheepfan0828, 2024-12-23:
> If you want to output logs in the console, check out the documentation:
> 
> https://prodocs.easyeda.com/en/api/guide/invoke-apis.html#entering-debug-mode
> 
> > add the `cll=debug` parameter within the URL of the EasyEDA Pro editor to enter debug mode.

And can you please solve the issue #14 first?

Sheepfan0828, 2024-12-24:
> If you want to output logs in the console, check out the documentation:
> 
> https://prodocs.easyeda.com/en/api/guide/invoke-apis.html#entering-debug-mode
> 
> > add the `cll=debug` parameter within the URL of the EasyEDA Pro editor to enter debug mode.

Hello! Just want to remind you to take a look on #14 and then #13.

easyeda2021, 2024-12-28:
最新版本2.2.34.8已经正式发布，该问题已经修复

closed 2024-12-28 by easyeda2021