easyeda/easyeda-api-sdk #11: Unable to get project info after switching document

radeeyate, 2024-12-18:
When using this code below, you can run it a couple times before switching to a different document, trying it again, and observing it not function.
```ts
import * as extensionConfig from '../extension.json';

// eslint-disable-next-line @typescript-eslint/no-unused-vars
export function activate(status?: 'onStartupFinished', arg?: string): void { }

export function about(): void {
  try {
    const boardInfo = eda.dmt_Project.getCurrentProjectInfo();
    Promise.resolve(boardInfo).then((info) => {
      if (info) {
        eda.sys_MessageBox.showInformationMessage(eda.sys_I18n.text(JSON.stringify(info)));
      } else {
        eda.sys_MessageBox.showInformationMessage("No board info available.");
      }
    });
  } catch (error) {
    eda.sys_MessageBox.showInformationMessage(String(error))
  }
}
```

The message box for errors is not even showing up, and instead I get the following errors in the console:
```
TypeError: Cannot read properties of undefined (reading 'attrsForSearch')
    at ui.js:2965:40305
    at dt (ui.js:1095:9647)
    at getCurrentProjectInfo (ui.js:2965:40232)
(anonymous) @ api.js:4
Promise.catch (async)
t @ api.js:4
execute @ api.js:4
push @ api.js:4
(anonymous) @ api.js:4
rpcCall @ api.js:4
getCurrentProjectInfo @ api.js:4
eval @ VM334:4
(anonymous) @ api.js:7
fr @ api.js:7
(anonymous) @ api.js:30
execute @ api.js:4
publish @ api.js:4
y @ ui.js:2172
onClick @ ui.js:2172
onClick @ ui.js:1101
vYi @ ui.js:1032
EYi @ ui.js:1032
jYi @ ui.js:1032
Gpt @ ui.js:1032
Wpt @ ui.js:1032
(anonymous) @ ui.js:1032
xje @ ui.js:1035
K1t @ ui.js:1032
xEe @ ui.js:1032
cEe @ ui.js:1032
ZYi @ ui.js:1032
preload.js:17 Uncaught Error: TypeError: Cannot read properties of undefined (reading 'attrsForSearch') https://modules.lceda.cn/pro-api/0.1.79.941a04f4/api.js 4 7059 Error: TypeError: Cannot read properties of undefined (reading 'attrsForSearch')
    at api.js:4:7059
    at rr.callback (api.js:4:6500)
    at rr.execute (api.js:4:6078)
    at zo.publish (api.js:4:7378)
    at api.js:4:6744
ir @ preload.js:17
api.js:4 Uncaught Error: TypeError: Cannot read properties of undefined (reading 'attrsForSearch')
    at api.js:4:7059
    at rr.callback (api.js:4:6500)
    at rr.execute (api.js:4:6078)
    at zo.publish (api.js:4:7378)
    at api.js:4:6744
(anonymous) @ api.js:4
(anonymous) @ api.js:4
execute @ api.js:4
publish @ api.js:4
(anonymous) @ api.js:4
PromiseRejectionEvent {isTrusted: true, reason: Error: TypeError: Cannot read properties of undefined (reading 'attrsForSearch')
    at https://mod…, type: 'unhandledrejection', target: Window, currentTarget: Window, …}
api.js:4 Uncaught (in promise) Error: TypeError: Cannot read properties of undefined (reading 'attrsForSearch')
    at api.js:4:7059
    at rr.callback (api.js:4:6500)
    at rr.execute (api.js:4:6078)
    at zo.publish (api.js:4:7378)
    at api.js:4:6744
```

yanranxiaoxi, 2024-12-19:
Thanks for the feedback.

After internal testing, it should be an interface parsing error due to data format modification. A bug ticket has been raised to fix it.

yanranxiaoxi, 2024-12-20:
It has been fixed within us. The fix will be released in the next beta version (2.2.34.6).

radeeyate, 2024-12-20:
Thanks for the quick response! Do you have any idea on when that might be released? And also - how might I be able to switch to the beta version?

yanranxiaoxi, 2024-12-20:
I'll let you know here when the version is available.

yanranxiaoxi, 2024-12-23:
当前的最新版本并不会上线公测。但你已经可以通过该地址访问它了：[https://pro.easyeda.com/2.2.34.6.9a154024/editor](https://pro.easyeda.com/2.2.34.6.9a154024/editor?cll=debug)

> [!WARNING]
> Please note: This is the address that specifies a fixed version (2.2.34.6), meaning that it will not be pushed to any newer version when it is in use.

radeeyate, 2024-12-23:
Great, thank you! My extension is now reporting the expected behavior.

radeeyate, 2024-12-26:
Hi @yanranxiaoxi,

Shortly after this banner was added this morning, the exact same error seems to be occuring even when on the new editor link.
[image]
[image]

easyeda2021, 2024-12-28:
you can download the 2.2.34.8 at https://easyeda.com/page/download

radeeyate, 2024-12-28:
The error is still occurring on 2.2.34.8. Is the extension API not up to date for that version?

Makerzy0226, 2025-04-08:
The issue you raised has been fixed. Please feel free to try it in the latest version (V2.2.37).

radeeyate, 2025-04-08:
Thank you! It's working well now!

closed 2025-04-08 by radeeyate