easyeda/easyeda-api-sdk #12: API社区论坛

Sheepfan0828, 2024-12-20:
你好 请问这个扩展api有没有其他的社区论坛呀？我想加入 有问题以获取更快的回复。

yanranxiaoxi, 2024-12-23:
当前没有在线支持社区。如果想要获得支持，在 GitHub 的 Issues 内最为合适。它可以让后来者能够查询到前人谈论过的内容以尽可能自主解决问题。

因为我们的开发资源有限，没有办法快速应答是正常的。我们希望尽可能以文档的形式来解决大多数人所能遇到的问题。

如果你觉得文档有什么内容没有写完善，可以在此提出，我们会尽可能优先地优化文档的内容。

-----

There is currently no online support community. If you want to get support, it's best to do so in the GitHub issue.

It allows latecomers to look up what has been talked about before and solve problems as autonomously as possible.

Because of our limited development resources, it's normal that there is no way to respond quickly. We want to be as documented as possible to solve the problems that most people can have.

If you feel that there is something in the document that is not well written, you can ask it here. We prioritize optimizing the content of the documentation as much as possible.

easyeda2021, 2024-12-28:
联系我们客服加入插件微信群

closed 2025-02-28 by yanranxiaoxi