easyeda/easyeda-api-sdk #13: 为器件设置位号时出错

Sheepfan0828, 2024-12-23:
你好 我通过`setState_Designator("D1")`来设置器件位号的时候出错了，以下是错误信息：
```
loader.js:2 
 Error: Get an illegal project!
    at x (pro-mgr.js:54:11550)
    at AsyncFunction.<anonymous> (pro-mgr.js:56:29971)
    at pro-mgr.js:55:1755
    at pro-mgr.js:56:8579
    at re.callback (loader.js:2:5157)
    at re.execute (loader.js:2:4084)
    at yt.publish (loader.js:2:4366)
    at loader.js:2:5510
    at new Promise (<anonymous>)
    at yt.rpcCall (loader.js:2:5399)
loader.js:2 
 Error: Get an illegal project!
    at x (pro-mgr.js:54:11550)
    at AsyncFunction.<anonymous> (pro-mgr.js:56:25279)
    at pro-mgr.js:55:1755
    at pro-mgr.js:56:8579
    at re.callback (loader.js:2:5157)
    at re.execute (loader.js:2:4084)
    at yt.publish (loader.js:2:4366)
    at loader.js:2:5510
    at new Promise (<anonymous>)
    at yt.rpcCall (loader.js:2:5399)
loader.js:2 
 Error: Get an illegal project!
    at x (pro-mgr.js:54:11550)
    at AsyncFunction.<anonymous> (pro-mgr.js:56:27777)
    at pro-mgr.js:55:1755
    at pro-mgr.js:56:8579
    at re.callback (loader.js:2:5157)
    at re.execute (loader.js:2:4084)
    at yt.publish (loader.js:2:4366)
    at loader.js:2:5510
    at new Promise (<anonymous>)
    at yt.rpcCall (loader.js:2:5399)
loader.js:2 
 Error: Get an illegal project!
    at x (pro-mgr.js:54:11550)
    at AsyncFunction.<anonymous> (pro-mgr.js:56:29971)
    at pro-mgr.js:55:1755
    at pro-mgr.js:56:8579
    at re.callback (loader.js:2:5157)
    at re.execute (loader.js:2:4084)
    at yt.publish (loader.js:2:4366)
    at loader.js:2:5510
    at new Promise (<anonymous>)
    at yt.rpcCall (loader.js:2:5399)
ui.js:2016 
 Uncaught (in promise) Error: Error: invalid projectId
    at Qnt.AddLibToProject (ui.js:2016:420032)
    at async r.placeDevice (ui.js:2016:468341)
    at async r.placeDevice (ui.js:3022:17384)
    at async ui.js:3022:16649
```
如下是我在`about`函数里面的代码实现：
```
  const library_uuid = await eda.lib_LibrariesList.getSystemLibraryUuid();
    if (!library_uuid) {
        throw new Error("Failed to get project library uuid!");
    }
    const comp_info = (await eda.lib_Device.search("SM4007PL_C64898", library_uuid)).shift();
    if (!comp_info) {
        throw new Error("Failed to create info");
    }
    const c_uuid = await comp_info.uuid;

    //console.log("libraryUuid: ", library_uuid, "uuid: ", c_uuid, "uuid from comp_info: ", await comp_info.libraryUuid);

    let comp = {libraryUuid: library_uuid, uuid: c_uuid};

    const c  = await eda.sch_PrimitiveComponent.create(comp, 0, 0);
    if (!c) {
        throw new Error("Failed to create ISCH_PrimitiveComponent.");
    }
    c.setState_Designator("D1");
    c.done();
    console.log("Component Created");
```
请问如何解决这个问题？

Sheepfan0828, 2024-12-24:
其中还有一个很奇怪的一点 我使用`setState_Designator`方法设置器件位号 但是画布中却显示在描述里面

Makerzy0226, 2025-04-08:
感谢反馈，这两个问题均已修复。

[image]